pimoroni/blinkt #1: Library is great, small suggestion

alexellis, 2016-06-22:
Hi Phil,

```
from blinkt import set_pixel, show
set_pixel(0,255,0,0)
show()
```

Would be great if this was like the scrollphat:

```
import blinkt
blinkt.set_pixel(0,255,0,0)
blinkt.show()
```

Super cool if class-based for testing / or swapping out different components like scroll phat or unicorn etc.

```
import blinkt
blink = blinkt()
blink.set_pixel(0,255,0,0)
blink.show()
```

Gadgetoid, 2016-06-22:
Thanks for your feedback!

Everything we hate about the class-based approach can be summed up with:

```
blink = blinkt()
```

Which is ugly! And redundant, since you can technically not use more than one Blinkt at once- okay by changing the pins and using some solid core wire you can use multiple, but you have to be trying pretty hard!

Maybe there's a way to compromise. I'm not sure!

Using:

```
blinkt.set_pixel(0,255,0,0)
```

vs

```
from blinkt import set_pixel
set_pixel(0,255,0,0)
```

Is generally a matter of preference. You can go either way. Just depends how much you like typing, or what other conflicting imports you might use. There's probably a page-long blog post somewhere arguing the benefits and drawbacks of each :D

alexellis, 2016-06-22:
I would be OK with the scrollphat/unicornhat style

closed 2016-07-06 by Gadgetoid

alexellis, 2016-07-08:
Closed because resolved?