pimoroni/blinkt #6: some remark

Wheele9, 2016-07-04:
The rainbow example is super cool.

In the mem_load.py you can use psutils.virtual_memory() instead of phymem_usage(). The new version of psutils changed the naming.

cpu_temp.py does not work with python 3

Gadgetoid, 2016-07-07:
Thanks for your feedback. Do you know when psutils changed, and if this is a Python 2 vs Python 3 divide? I'd be inclined to do some sort of check for "virtual_memory" with a fallback to "phymem_usage" or something to that effect. Although the naming conventions make these sound like totally different things. Might need to do some reading ;)

Wheele9, 2016-07-07:
psutils changed from version 0.6: https://mail.python.org/pipermail/python-announce-list/2012-August/009575.html

> === Memory functions ===
> 
> psutil.phymem_usage() and psutil.virtmem_usage() are deprecated.
> Instead we now have psutil.virtual_memory() and psutil.swap_memory(),
> which should provide all the necessary pieces to monitor the actual
> system memory usage, both physical and swap/disk related.

 I think its not a 2 vs 3 divide, I tried with both of them and "phymem_usage" failed always, "virtual_memory" succeded always

commit 2892eb08 (2016-07-08): Changed psutil.phymem_usage to psutil.virtual_memory. Closes #6.
closed 2016-07-08 by sandyjmacdonald