pimoroni/blinkt #9: Node.js Implementation

Irrelon, 2016-07-09:
Hi folks,

I've been trying to put together a Node.js version of this using the wiring-pi GPIO library.

Could you give me some hints on how the changes to the GPIO pins actually control the colour and brightness of each pixel?

For instance, I can set pin 23 and 24 with wiringpi's digitalWrite(pin, value) method but what is the sequence blinkt is expecting in order to get it to change colours etc?

Many thanks!

Irrelon, 2016-07-09:
Scratch that, I have managed to convert the python blinkt.py control methods to javascript methods in Node.js and they are working :)

I will share once I've packaged them up into something worth publishing rather than hacky code :)

closed 2016-07-09 by Irrelon

Irrelon, 2016-07-09:
Further to this, I have produced a Node.js version. You can find it here: https://www.npmjs.com/package/node-blinkt

If you wish, you can update your readme to show this link for Node.js users. The license is any open-source e.g. MIT, GPL etc.

neauoire, 2018-04-09:
Thank you so much for this.