pimoroni/blinkt #13: No modue named numpy on Jessie Lite

alexellis, 2016-07-22:
```
pi@blinkt1:~/Pimoroni/blinkt $ python examples/pulse.py
Traceback (most recent call last):
  File "examples/pulse.py", line 5, in <module>
    import numpy as np
ImportError: No module named numpy
```

**Suggest adding** `pip install numpy` or apt-get if it exists to the `curl | sh` command

alexellis, 2016-07-22:
Output from the installation is below. This felt like it took quite a long time on my Pi Zero. There is a second y/N question but I don't think I got a prompt from the script. 

```
pi@blinkt1:~ $ curl -sS get.pimoroni.com/blinkt | bash

This script will install everything needed to use
Blinkt!

--- Warning ---

Always be careful when running scripts and commands
copied from the internet. Ensure they are from a
trusted source.

If you want to see what this script does before
running it, you should run:
    \curl -sS get.pimoroni.com/blinkt

Do you wish to continue? [y/N] y

Checking hardware requirements...
Done!

Checking install requirements...

Updating package indexes...
Get:1 http://archive.raspberrypi.org jessie InRelease [13.2 kB]
Get:2 http://mirrordirector.raspbian.org jessie InRelease [14.9 kB]
Get:3 http://archive.raspberrypi.org jessie/main armhf Packages [144 kB]
Get:4 http://archive.raspberrypi.org jessie/ui armhf Packages [8,966 B]
Get:5 http://mirrordirector.raspbian.org jessie/main armhf Packages [8,981 kB]
Ign http://archive.raspberrypi.org jessie/main Translation-en_GB
Ign http://archive.raspberrypi.org jessie/main Translation-en
Ign http://archive.raspberrypi.org jessie/ui Translation-en_GB
Ign http://archive.raspberrypi.org jessie/ui Translation-en
Get:6 http://mirrordirector.raspbian.org jessie/contrib armhf Packages [37.5 kB]
Get:7 http://mirrordirector.raspbian.org jessie/non-free armhf Packages [70.3 kB]
Get:8 http://mirrordirector.raspbian.org jessie/rpi armhf Packages [1,356 B]
Ign http://mirrordirector.raspbian.org jessie/contrib Translation-en_GB
Ign http://mirrordirector.raspbian.org jessie/contrib Translation-en
Ign http://mirrordirector.raspbian.org jessie/main Translation-en_GB
Ign http://mirrordirector.raspbian.org jessie/main Translation-en
Ign http://mirrordirector.raspbian.org jessie/non-free Translation-en_GB
Ign http://mirrordirector.raspbian.org jessie/non-free Translation-en
Ign http://mirrordirector.raspbian.org jessie/rpi Translation-en_GB
Ign http://mirrordirector.raspbian.org jessie/rpi Translation-en
Fetched 9,272 kB in 26s (352 kB/s)
Reading package lists... Done

Checking for dependencies...
Checking for git...
git required. Installing.

Reading package lists... Done
Building dependency tree
Reading state information... Done
The following extra packages will be installed:
  git-man libcurl3-gnutls liberror-perl rsync
Suggested packages:
  gettext-base git-daemon-run git-daemon-sysvinit git-doc git-el git-email git-gui gitk gitweb git-arch git-cvs git-mediawiki git-svn
The following NEW packages will be installed:
  git git-man libcurl3-gnutls liberror-perl rsync
0 upgraded, 5 newly installed, 0 to remove and 28 not upgraded.
Need to get 4,613 kB of archives.
After this operation, 21.5 MB of additional disk space will be used.
Get:1 http://mirrordirector.raspbian.org/raspbian/ jessie/main libcurl3-gnutls armhf 7.38.0-4+deb8u3 [225 kB]
Get:2 http://mirrordirector.raspbian.org/raspbian/ jessie/main liberror-perl all 0.17-1.1 [22.4 kB]
Get:3 http://mirrordirector.raspbian.org/raspbian/ jessie/main git-man all 1:2.1.4-2.1+deb8u2 [1,267 kB]
Get:4 http://mirrordirector.raspbian.org/raspbian/ jessie/main git armhf 1:2.1.4-2.1+deb8u2 [2,725 kB]
Get:5 http://mirrordirector.raspbian.org/raspbian/ jessie/main rsync armhf 3.1.1-3 [373 kB]
[... 374 lines cut ...]
Downloading examples and documentation...

Cloning into 'blinkt'...
remote: Counting objects: 65, done.
remote: Compressing objects: 100% (56/56), done.
remote: Total 65 (delta 4), reused 43 (delta 2), pack-reused 0
Unpacking objects: 100% (65/65), done.
Checking connectivity... done.
Resources copied to /home/pi/Pimoroni/blinkt/

Checking for additional software...

Some Blinkt! examples/plugins
require additional software and/or modules
Would you like to install them now? [y/N]
Finalising Install...

Reading package lists... Done
Building dependency tree
Reading state information... Done

All done!

Enjoy your Blinkt!!
```

RogueM, 2016-07-23:
numpy has been added to the OLI, but, yes, being a requirement from a specific example, you would need to opt-in their installation.

since examples without those supporting modules don't make a lot of sense, I will look into making the opt-in/out for the examples themselves in the future, so users can't end up with one without the other... well, using the OLI anyhow.

dglaude, 2016-07-24:
Maybe you should split example into those without additional requirement and those with.
There was a split like that in an other pimoroni repository, maybe Unicorn HAT.
So user wanting demo but not installation of additional module can get the best of both world.

RogueM, 2016-07-24:
... I think you are referring to the enviro-phat repo? if so, the separation there exists because other add-on board (and related lib) are necessary and separate installation are involved.

I guess, ultimately, our expectation is that the one-line installers will be used to install lib, doc and examples in most cases... those installing from github should be savvy enough to troubleshoot possible issues doing so.

Going forward, hopefully, it will be less of an issue as all examples now include a 'friendly' message telling people how to install additional modules (which can be less than straightforward when the PyPi module name doesn't match the Python interpreter output).

Still, there will be commits to the repos introducing new examples that don't take great care in the module import handling, so I know every now and then there will be issue like this popping up, but I want to move the problem to the repo rather than the installer. Or at least minimise the need to be reactive outside the repo.

alexellis, 2016-07-24:
How big can numpy be? I would think it is worth putting all extra libraries in the n/Y question at the end.

dglaude, 2016-07-24:
Did not check enviro-phat repo.
What you have in Unicorn-hat is a requirement statement for example that require additional libs.
The real difficulty for the developper is that he may have many libs installed from another project. What I try to do is reinstall a jessy-lite from scratch and only install one HAT/pHAT related software to see if it works without addition. But it take time to do it right.

About the one liner installer... don't you want to make it two liner and explanation on security issues? Like only download script you run from known source and over https.
It is really a frightening practice to see all those blind installation, sometime over HTTP.

RogueM, 2016-07-24:
@alexellis , as stated... numpy is now included, but @Gadgetoid is in control of pushing them... and rightfully 'enjoying' the sun ;-)

@dglaude yes, that is exactly what we are doing for all repos. I really don't think it is difficult for a dev to know what is standard lib, or addition... the can just look up the official module index:

https://docs.python.org/2/py-modindex.html
or
https://docs.python.org/3/py-modindex.html

... that said, yes, I don't expect all submission to be spot-on, it's unrealistic. So, we'll just aim to review commits of examples and make sure we adjust the import section if needs be.

note that re OLI, there is a dedicated repo: http://github.com/pimoroni/get
for any suggestion or concerns, it is best to take it there.

closed 2016-08-26 by Gadgetoid