pimoroni/blinkt #14: cpu_load.py No module named psutil

alexellis, 2016-07-22:
```
pi@blinkt1:~/Pimoroni/blinkt/examples $ python cpu_load.py
Traceback (most recent call last):
  File "cpu_load.py", line 6, in <module>
    import psutil
ImportError: No module named psutil
```

alexellis, 2016-07-22:
```
Cpi@blinkt1:~/Pimoroni/blinkt/examples $ python mem_load.py
Traceback (most recent call last):
  File "mem_load.py", line 6, in <module>
    import psutil
ImportError: No module named psutil
```

RogueM, 2016-07-23:
examples require extra modules, on an ad hoc basis. I have submitted PR #15 that should somewhat makes the process more friendly.

closed 2016-08-11 by Gadgetoid

wmurray-es, 2020-01-14:
Still seems to be an issue to this day when using the `sudo apt-get install pimoroni -y` method