pimoroni/blinkt #26: Cannot get it to work (2 of them)

PJOttCan, 2016-08-18:
I bought a RPiZero to try this with.  And I purchased two Blinkt! items for redundancy (neither work). I enabled Remote GPIO in RPi Config but none of the examples are working.  It is situated correctly and I'm using the 1 x Dual row, 40 pin GPIO header to connect it to the Zero.  Is there a diagnostic/self-test the Zero can do for GPIO without connecting anything to it?  Am I using the wrong orientation of the header? (short vs long end?)  I'm not an electronics nerd but I am a programming nerd so was hoping to devise some neat things to do with Blinkt!

RogueM, 2016-08-18:
It's probably not the best place to try to troubleshoot your problem. Best would be to open a topic on our forums: http://forums.pimoroni.com/c/support

PJOttCan, 2016-08-24:
I figured it out.

closed 2016-08-24 by PJOttCan

Taru-garg, 2019-12-26:
Hey!! can you tell me what did you do to get them working