pimoroni/blinkt #28: Install error

GrovesDashSystemsDotCom, 2016-09-01:
Hi. Pi Zero running Raspian 8 with latests updates.

Running install script (with and without sudo) (Y to all options asked) gives the below output.

Any guidance much appreciated.

Thanks!

Jonathan

_Installing Blinkt! library for Python 2...

Downloading/unpacking blanket
  Downloading blinkt-0.0.7.tar.gz
  Running setup.py (path:/tmp/pip-build-CBbVSW/blinkt/setup.py) egg_info for package blanket

Downloading/unpacking RPi.GPIO (from blanket
  Downloading RPi.GPIO-0.6.2.tar.gz
  Running setup.py (path:/tmp/pip-build-CBbVSW/RPi.GPIO/setup.py) egg_info for package RPi.GPIO

Installing collected packages: blinkt, RPi.GPIO
  Running setup.py install for blanket

  Running setup.py install for RPi.GPIO
    building 'RPi._GPIO' extension
    arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -I/usr/include/python2.7 -c source/py_gpio.c -o build/temp.linux-armv6l-2.7/source/py_gpio.o
    source/py_gpio.c:23:20: fatal error: Python.h: No such file or directory
     #include "Python.h"
                        ^
    compilation terminated.
    error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
    Complete output from command /usr/bin/python -c "import setuptools, tokenize;__file__='/tmp/pip-build-CBbVSW/RPi.GPIO/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-jhLEww-record/install-record.txt --single-version-externally-managed --compile:
    running install

running build

running build_py

creating build

creating build/lib.linux-armv6l-2.7

creating build/lib.linux-armv6l-2.7/RPi

copying RPi/__init__.py -> build/lib.linux-armv6l-2.7/RPi

creating build/lib.linux-armv6l-2.7/RPi/GPIO

copying RPi/GPIO/__init__.py -> build/lib.linux-armv6l-2.7/RPi/GPIO

running build_ext

building 'RPi._GPIO' extension

creating build/temp.linux-armv6l-2.7

creating build/temp.linux-armv6l-2.7/source

arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -I/usr/include/python2.7 -c source/py_gpio.c -o build/temp.linux-armv6l-2.7/source/py_gpio.o

source/py_gpio.c:23:20: fatal error: Python.h: No such file or directory

 #include "Python.h"

                    ^

compilation terminated.

error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1

----------------------------------------
Cleaning up...
Command /usr/bin/python -c "import setuptools, tokenize;__file__='/tmp/pip-build-CBbVSW/RPi.GPIO/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-jhLEww-record/install-record.txt --single-version-externally-managed --compile failed with error code 1 in /tmp/pip-build-CBbVSW/RPi.GPIO
Storing debug log for failure in /root/.pip/pip.log
Python 2 library install failed!
If problems persist, visit forums.pimoroni.com for support_)

AndiCui, 2016-09-01:
happened to me as well 
try this
```
sudo apt-get update
sudo apt-get install python-dev python3-dev
```

GrovesDashSystemsDotCom, 2016-09-01:
Trying now, thanks!

J

GrovesDashSystemsDotCom, 2016-09-02:
All good now, thanks!

closed 2016-09-02 by GrovesDashSystemsDotCom