pimoroni/blinkt #30: More info for installing and working with the blinkt

dstinebaugh, 2016-09-15:
I'd personally love to get it working on kali linux but of course the install script only works for raspbian, I have wiring pi installed and assume I should be able to do some pwm using it instead of the full blinkt library, if I knew what pin's did what ect. or at least a list of dependancies so I can try to install this without tearing down the install script and trying to reverse it.

RogueM, 2016-09-15:
The blinkt script is actually one out of a handful we've recently not restricted to Raspbian. I haven't tried it on Kali yet, so it may still be restricted though, I'll check it out in the next few days.

In term of pin used, those can be found at pinout.xyz, or checking out the livrary itself, where they are defined by necessity. Checking out the code should also help you if you want to try to control blonkt with wiringpi... I would think that is possible indeed.

RogueM, 2016-09-15:
right, I had a quick glance and I think the problem is 2-fold:

1) Kali will not currently pass the test of 'supported' releases that the script filters in (that could be easily allowed).

2) More of a problem is the fact that the default user out of the box is the super-user... it would be a pretty big (ir)responsibility to open the scripts to Kali on those basis. So at the very least a non-root user with sudo rights should be available.

I'll check tomorrow in more details, but in the meantime the following should hopefully be helpful:

http://dirkwillem.nl/apa102-led-strip/

> The APA102 works by sending bytes through the SPI interface. It is started by sending 32 bits containing 0, then a 32-bit frame for every LED in the strip, and then 32 bits containing 1. Every LED frame contains a constant part (binary 111), 5 bits containing the brightness of the LED, and 3 8-bit integers containing the blue, green and red components respectively.

dstinebaugh, 2016-09-16:
Wasn't aware of pinout.xyz that's a handy resource thanks! I also wanted to say thanks for the current install script and library I didn't want it to sound like I don't enjoy what you have already provided, I've just picked up some blinkt's and  mashed the random and cheerlights example together and have really enjoyed it so far!

As for Kali I do have a non root user so that's an easy problem to overcome, and have tried changing the os check from raspbian to kali just to see how far it would get, looks like python-rpi.gpio isn't found but the rest completes without issue so it shouldn't be to much more work to get the full library installed, a fun project for this weekend.

Thanks again RogueM for chiming in!

RogueM, 2016-09-16:
Yes, you are right python-rpi.gpio is not available in the kali repo, however, after apt fails to install that particular library it should fulfil the dependency using pypi. In other words, I expect you to have a fully working install regardless?

I did a little test last night and blinkt should not be a massive problem to open up. Others would be largely more complicated and they expect some config files that Kali does not provide, which may be a show stopper for those.

RogueM, 2016-09-16:
There you go... we've uploaded a new version of the script relaxed fot Kali.

It will complain about python-rpi.gpio not being available in the base repo but since this is a dependency specified at the lib level, pypi will kick in. I'm working on a smarter (or rather less alarming for users) way to handle deb that are specific to thr raspberry pi such as rpi.gpio, but I need to move more cautiously as that is part of a larger picture of multi-os support.

Anyhow, all the examples dependencies seem to be fulfil in the Kali repo, so just running the script should give you roughly the same experience as Raspbian users... let me know if any problems!

closed 2017-02-06 by Gadgetoid