pimoroni/blinkt #31: All example should be like 1D Tetris

dglaude, 2016-09-21:
I mean with this kind of code:
if __name__ == "__main__":
    main()

The idea is that in microPython, one can not easily exec a python script. What we do is import, but you can not unimportant, so you can only run once. With mail() trick, one can import and run as many time as wanted.

I have other microPython related issue, such as time resolution that is only integer. But I am not sure if there is an alternative working on both versions of Python.

Gadgetoid, 2017-02-06:
I'm guessing integer time resolution coupled with any script that uses `sin()` to create time-based oscillation effects (bouncing through the hue in the HSV colour space for example) makes for really nasty stepping in the current scripts?

What's the resolution of time in Miro Python? Is it microseconds, milliseconds? What's the integer size?

Gadgetoid, 2017-02-06:
Oh and I don't disagree with the `__name__ == "__main__"` pattern necessarily, it's certainly something we should consider since one way or another all the examples should be consistent and in line with best practise.

dglaude, 2017-02-07:
It is not fair to reply so long after my message. It is so last year, maybe
another epoch. I don't remember much about my ESP8266 period.

Anyway, the issue with utime.time is the second resolution. From the doc:
If you want to develop portable MicroPython application, you should not
rely on this function to provide higher than second precision. If you need
higher precision, use ticks_ms() and ticks_us() functions

And because ticks warp around you have to use utime.ticks_diff(*ticks1*,
*ticks2*) to compare.

I don't know how to have portable code that work with python on the Pi and
microPython on ESP8266 as the name exist in both but with different
behaviour. I had to modify the example to make them work.

David GLAUDE

closed 2017-03-22 by Gadgetoid