pimoroni/blinkt #33: Create a VU meter

francoismartin, 2016-09-28:
Hi,

Do you have idea, how can I create a vu meter with blinkit with the input and output sound ?
Like that : https://vimeo.com/114906251 (for output sound)

Thanks

Gadgetoid, 2017-01-24:
It's probably not as helpful as it could be, but for Speaker pHAT we've done this using an ALSA plugin: https://github.com/pimoroni/speaker-phat/tree/master/alsa-plugin

I will be adding APA102 support into this code at some point, which may include Blinkt support simply because that would be cool :D

Gadgetoid, 2017-02-06:
This has now been moved to its own project, Pi VU Meter, which includes Blinkt support specifically: https://github.com/pimoroni/pivumeter

It's a little embryonic at the moment, but we'll be building a nice easy to use package eventually.

closed 2017-02-08 by Gadgetoid