pimoroni/blinkt #35: Brightness param appears to have no effect

C-D-Lewis, 2016-10-09:
When I use the brightness parameter in set_pixel (as well as by set_brightness), the LEDs are still extremely bright. Could this be something to do with a pin other than 5V, GND, DATA, and CLOCK? I soldered only these four pins due to this Pi Zero being used only for this hat. 

Example code:
```
blinkt.set_pixel(0, 0, 255, 0, BRIGHTNESS)
blinkt.show()
```

RogueM, 2016-10-09:
Right, but what value are you passing for the BRIGHTNESS parameter, your code is not explicit in that respect. I believe any value out of bound, namely over 1.0 will be capped, not rejected (code failing).

C-D-Lewis, 2016-10-09:
@RogueM Thanks for the swift response. Apologies, I pasted the least helpful line... I have tried values between 0 and 1, such as 0.5, 0.3, etc.

RogueM, 2016-10-09:
what is the library version, i.e if you check with:
`pip list`

C-D-Lewis, 2016-10-09:
That command lists the version as `blinkt (0.1.0)`. I installed via the `curl -sS get.pimoroni.com/blinkt | bash` command.

RogueM, 2016-10-09:
I'll check it out to be sure it works as expected. To answer your earlier question, no, that's the only pins required (in fact they are the only electrically connected pins on blinkt).

C-D-Lewis, 2016-10-09:
Thanks, that is good information to know. Let me know what you find. It's a PI Zero, if that makes any difference.

RogueM, 2016-10-10:
I can confirm it works as intended, both as part of a set_pixels call or set_brightness(). Try to change brightness jumping from 0.1 and 1 or vice versa and it should be quite visibly different. Between 0.3 and 0.5 it is quite a bit more subttle.

C-D-Lewis, 2016-10-10:
After more experimentation, it does indeed appear that the param works, I was just expecting there to be a larger dynamic range. You may close this issue, and thanks for your time.

Gadgetoid, 2017-02-06:
As a closing note- the Blinkt brightness setting is actually changing the 5-bit value set to the pixels. As such it only has valid values between 0 and 31, not much in the way of range. It's also not gamma corrected, so brightness changes will be a lot more obvious at the low end of the scale- as @RogueM points out.

It can, however, be used in conjunction with scaling the R/G/B values to produce a wider dynamic range. a crude method would be something like:

```
brightness = 0.5
r *= brightness
g *= brightness
b *= brightness

blinkt.set_pixel(0, r, g, b,  1.0)
```

But you could scale both colour channel brightness and overall brightness together to potentially better effect.

closed 2017-02-06 by Gadgetoid

C-D-Lewis, 2017-02-06:
@Gadgetoid Thanks for the insight, I have in fact been successfully using this library by scaling the rgb values as you suggested for the last few months! Long live Blinkt!