pimoroni/blinkt #45: Unable to locate package python3-tweepy

jonwitts, 2017-02-26:
Hi there, when installing via the "curl -sS https://get.pimoroni.com/blinkt | bash" method and performing a full install I get the following error...

```
Installing python3-tweepy...
E: Unable to locate package python3-tweepy
Apt failed to install python3-tweepy!
Falling back on pypi...
Traceback (most recent call last):
  File "/usr/bin/pip3", line 9, in <module>
    load_entry_point('pip==1.5.6', 'console_scripts', 'pip3')()
  File "/usr/lib/python3/dist-packages/pkg_resources.py", line 356, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python3/dist-packages/pkg_resources.py", line 2476, in load_entry_point
    return ep.load()
  File "/usr/lib/python3/dist-packages/pkg_resources.py", line 2190, in load
    ['__name__'])
  File "/usr/lib/python3/dist-packages/pip/__init__.py", line 74, in <module>
    from pip.vcs import git, mercurial, subversion, bazaar  # noqa
  File "/usr/lib/python3/dist-packages/pip/vcs/mercurial.py", line 9, in <module>
    from pip.download import path_to_url
  File "/usr/lib/python3/dist-packages/pip/download.py", line 25, in <module>
    from requests.compat import IncompleteRead
ImportError: cannot import name 'IncompleteRead'
```
Thanks,
Jon

RogueM, 2017-02-26:
that means `pip3 install tweepy` cannot run, and from experience I'd say it's because pip3 itself is broken.

jonwitts, 2017-02-26:
But it says that it is apt which cannot find it, not pip3...

RogueM, 2017-02-27:
right, it checks `apt` first than falls back on `pypi`... try it!

Gadgetoid, 2017-03-17:
```
ImportError: cannot import name 'IncompleteRead'
```

I wish I could remember how to fix this from any day to the other. I've done it 3-4 times now. Think it usually involves a full uninstall and reinstall of python-pip via apt.

RogueM, 2017-03-17:
I believe that is the nightmare required yes... my rule of thumb is, don't `pip install` a requirement file until you've checked `requests` in not in the list... it has kept me dry and with my pants on since I have religiously adopted that mantra!

closed 2017-04-12 by Gadgetoid