pimoroni/blinkt #46: Sample not working on Pi Zero W

WebberLai, 2017-03-17:
Hi , I follow the instructions .

After I ssh my pi zero , I enter cmd

`curl -sS https://get.pimoroni.com/blinkt | bash`

than 

`cd /home/pi/Pimoroni/blinkt/examples`

I see many python sample inside the folder

When I try to run the **random_blink.py** or **random_blink_colours.py**

Nothing happen after I type `sudo python random_blink.py`

Why ??? Did miss anything ???

WebberLai, 2017-03-17:
LED position is upside down ... I did not notice it has direction

closed 2017-03-17 by WebberLai

RogueM, 2017-03-17:
on a somewhat unrelated subject, note that you don't have to run the command above with `sudo`... take the habit not to use `sudo` (elevated privileges) unless necessary / specifically instructed to do so.