pimoroni/blinkt #56: Bug in larson.py

bablokb, 2017-07-15:
In  examples/larson.py you will run into an `IndexError` if delta is an exact multiple of 16:

    offset = int(abs((delta % 16) - 8))
    for i in range(8):
      blinkt.set_pixel(i , REDS[offset + i], 0, 0)

The list `REDS` only has 15 values (max index is 14), but if offset is 8 you are requesting index 15.

Bernhard

dglaude, 2017-07-15:
I may have discover this bug while porting larson to mote stick.
My souvenir is indeed that you need to grow the array a little bit.

Gadgetoid, 2017-07-18:
Yikes. Good spot. It should be `delta % len(REDS)` ideally.

closed 2017-07-19 by Gadgetoid

bablokb, 2017-07-19:
Hm, just a bit of nitpicking: you still have a magic number in your code:

    delta = (time.time() - start_time) * 16
        offset = int(abs((delta % len(REDS)) - blinkt.NUM_PIXELS))
couldn't you also replace the 16 with len(REDS)?

Anyhow, it's just an example and I think you are doing a wonderful job giving us users a headstart on implementing our own solutions.

Bernhard