pimoroni/blinkt #57: Install one-liner not fetching up-to-date examples

ncguk, 2017-08-07:
I've just installed the Blinkt software using the `curl https://get.pimoroni.com/blinkt | bash` one-liner (on a Pi Zero W, Raspbian Jessie, with everything up to date), and have noticed that there are fewer examples in the resulting 'full install' than there are in the main branch of the GitHub repo.

Changing `copyhead="no"` to `copyhead="yes"` in the installer bash script solves this, but I'm guessing there is a better solution than that.

Gadgetoid, 2017-08-08:
This is intentional, so that examples relying on newer and potentially unreleased library features are not copied down with a version of the library that doesn't support them. Sometimes we'll, as you noticed, bump to `copyhead="yes"` because we know the examples will work.

ncguk, 2017-08-08:
Okay, gotcha. I'll close the issue.

closed 2017-08-08 by ncguk