pimoroni/blinkt #71: python3-blinkt package hasn't installed python3 module

martinohanlon, 2018-12-30:
I have run the blinkt installation and apt reports that the `python3-blinkt` package has been installed but `blinkt` cannot be imported by python 3.

[image]

martinohanlon, 2018-12-30:
Fyi - I had to install `blinkt` using `pip3` to get it to work.

[image]

Gadgetoid, 2019-02-21:
I see this from time to time and it often indicates a corrupt package install- I don't know if that was the case in this specific instance. Glad you're up and running though.

closed 2019-02-21 by Gadgetoid