pimoroni/blinkt #79: Unable to Run on an RPi Zero with Buster

ryancasler, 2019-11-13:
After installing on an RPi Zero W on a clean install of Raspbian Buster, I'm unable to run any of the examples.  It looks like something should be happening but none of the lights actually come on.  Any idea what I could be doing wrong or what else has to be done in Raspbian Buster to get these to run?

Gadgetoid, 2019-11-14:
I'm not aware of any problems with Buster, you should:

1. Make sure you don't have anything else running that could be interfering with the pins Blinkt! uses (https://pinout.xyz/pinout/blinkt)
2. Make sure you've got the board on the header the right way around (we've all been there)
3. Make sure it's pushed down firmly on the header

ryancasler, 2019-11-14:
It was a fresh install of Buster on Raspbian so I know there is nothing else conflicting.  The board is seated well and facing the correct direction.  Is there anything else other than installing the python library that has to be done in order to get it to work?  The only thing in the instructions is to run the 
`curl https://get.pimoroni.com/blinkt | bash`  command.

Gadgetoid, 2020-02-25:
Sorry for the delay- did you ever get to the bottom of this?

ryancasler, 2020-02-27:
No.  I was never able to get it to work.

Gadgetoid, 2020-03-16:
It could be an issue with the Pi/Blinkt! Might be worth contacting our support (support@pimoroni.com) if you haven't already, since they're more used to dealing with this sort of thing than me.

closed 2020-03-16 by Gadgetoid