pimoroni/blinkt #80: Add support for dotstar LED strips

tiziano88, 2020-01-23:
[Dotstar LED strips](https://shop.pimoroni.com/products/flexible-rgb-led-strip-dotstar-apa102-sk9822-compatible) are compatible with APA102, and I found I was able to use this library to drive them, by just tweaking some hardcoded constants such as the number of pixels (in fact that may be the only one):

https://github.com/pimoroni/blinkt/blob/ac1fda55fbde31cb243aa43bd7a9b779c32d6fae/library/blinkt.py#L12-L15

It would be nice if the library had a better encapsulated mechanism to initialise itself, instead of using global mutable variables, e.g. it may return an object that holds the number of LEDs and their state, and expose methods to modify it. That said, I am not really a Python person, so I am not sure what is idiomatic.

Gadgetoid, 2020-01-23:
This library is intended to be more beginner-friendly and highly specific to our Blinkt! add-on board, but what you suggest is very relevant for the direction I'd like to take another project of ours in, plasma - https://github.com/pimoroni/plasma

With plasma examples you can supply your choice of GPIO pins on the command line as detailed in the comments here - https://github.com/pimoroni/plasma/blob/master/examples/rainbow.py

It's a bit of a shift for a library that was originally intended for driving illuminated arcade buttons, but I think it makes sense and your input would be welcome!

Gadgetoid, 2020-07-13:
You can now find a minimal library for APA102 pixels here - https://github.com/pimoroni/apa102-python

closed 2023-11-02 by Gadgetoid