pimoroni/blinkt #81: Issues out of the box with psutil (cpu_load.py)

metadiablo-glyph, 2020-04-12:
```
pi@raspberrypi:~ $ sudo python3 cpu_load.py
This script requires the psutil module
Install with: sudo pip install psutil
pi@raspberrypi:~ $ sudo pip install psutil
sudo: pip: command not found
```

This is after using `sudo apt install python3-blinkt` on the latest Raspbian Buster Lite (fresh install and update/upgrade)

> Raspbian Buster Lite
> Version: February 2020
> Release date: 2020-02-13
> Kernel version: 4.19

Similar: https://github.com/pimoroni/blinkt/issues/14

I imagine this is not intended functionality. It would be much friendlier to have it all work out of the box. (or have instructions to allow this to be easier)

druck13, 2020-04-12:
On 12 Apr 2020 Glyph <notifications@github.com> wrote:

> ```
> pi@raspberrypi:~ $ sudo python3 cpu_load.py
> This script requires the psutil module
> Install with: sudo pip install psutil
> pi@raspberrypi:~ $ sudo pip install psutil
> sudo: pip: command not found

Try pip3 rather than pip, as pip is for python 2.

Otherwise do a:
     sudo apt install python3-pip

> I imagine this is not intended functionality. It would be much friendlier
> to have it all work out of the box. (or have instructions to allow this to
> be easier)

Before releasing any software it a good idea to try it on a completely 
fresh install of the OS. Its very easy to forget the things that you 
always install yourself, but other might not have. It's also something I 
forget to do on a regular basis.

Cheers
---David

-- 
______________________________________________________________________

David J. Ruck   Phone: +44- (0)7974 108301   Email: druck@druck.org.uk
______________________________________________________________________

metadiablo-glyph, 2020-04-12:
pip3 returns the same error as I don't have pip/pip3 installed on a fresh Raspbian image. These tests were done on a completely fresh install of Raspbian Buster Lite, with update/dist-upgrade, however this is definitely sage advice!

There are no instructions that layout install instructions for pip/pip3 in the readme, of course with sudo apt install python3-pip this changes. The readme for a manual install (which is of course required on Lite images) Simply says to use `sudo apt-get install python3-blinkt` Which alone does not solve the issues of three problems, the first being the examples dependencies are not met, the second is that for the _uninitiated_ the pip/pip3 install is not very obvious, and finally the Development install instructions do not work as stated.

Customers who purchased this hardware may find it rather frustrating having to go through hoops to still use it, and those buying these things aren't likely very experienced with Raspbian in general. Leaving a bad taste for folks trying to get into "making" after buying products from Pimoroni.

RogueM, 2020-04-13:
I recommend you use the pimoroni-dashboard (sudo apt-get install pimoroni) or the dedicated scripts for the blinkt as they take care of the necessary dependencies, not just for the library, but the examples. It does not mean they will go a perfect job on Buster but they will definitely get yoiu further than a manual install, if you are unsure how to proceed.

RogueM, 2020-04-13:
the dedicated script is mentioned here:
`https://learn.pimoroni.com/tutorial/tanya/beginning-with-blinkt`

... but again, I recommend using the `pimoroni-dashboard`, which is just a 'smart' frontend for these scripts. Note that Buster has changed a few things so install should come with a health hazard, and may require some additional massaging.

RogueM, 2020-04-13:
oner thing I will say on Buster though, is that you will need to run the following manually first, preferably: `sudo apt-get raspi-gpio install`

but other than that the instructions set out on the front page works as advertised as far as I can tell. Development section is obviously for advanced users and require additional knowledge.

metadiablo-glyph, 2020-04-13:
> Full install (recommended):
> 
> We've created an easy installation script that will install all pre-requisites and get your Blinkt! up and running with minimal efforts. To run it, fire up Terminal which you'll find in Menu -> Accessories -> Terminal on your Raspberry Pi desktop, as illustrated below:

This leads me to believe it only works with "Raspbian with Desktop".  I am certainly no expert but it's misinterpretable. I thought the `curl | bash` method only applied to folks with Raspbian w/ Desktop as I felt it was a pretty big context clue.

Gadgetoid, 2020-04-14:
Our Python boilerplate seeks to fix from-GitHub install issues with Raspbian Lite (road tested with Enviro+), but those improvements have yet to be backported to Blinkt!

I have a process document for revisiting our old software libraries - bear in mind this one is 3-4 years old now - and Blinkt! will get its turn sooner or later.

These changes include documenting apt dependencies, which are still an unfortunate reality of shipping Python on the Pi - https://github.com/pimoroni/boilerplate-python/blob/f4a64c221efdb358bd78bf5b56bff323cca4ac5f/library/setup.cfg#L45-L49

closed 2020-04-15 by metadiablo-glyph