pimoroni/blinkt #85: RuntimeError: Not running on a RPi! on DietPi

gogorz, 2020-05-09:
The title pretty much. I use DietPi and it shows this error. Raspberry Pi 4B+.

```
dietpi@raspi:~/Pimoroni/blinkt/examples$ python3 cpu_load.py 
Traceback (most recent call last):
  File "cpu_load.py", line 31, in <module>
    show_graph(v, 255, 255, 255)
  File "cpu_load.py", line 25, in show_graph
    blinkt.show()
  File "/usr/lib/python3/dist-packages/blinkt.py", line 76, in show
    GPIO.setup(DAT, GPIO.OUT)
RuntimeError: Not running on a RPi!
```

druck13, 2020-05-10:
On 09/05/2020 11:28, gogotz wrote:
> The title pretty much. I use DietPi and it shows this error. Raspberry
> Pi 4B+.
>
> |dietpi@raspi:~/Pimoroni/blinkt/examples$ python3 cpu_load.py Traceback
> (most recent call last): File "cpu_load.py", [...]

This might help https://github.com/adafruit/Adafruit_Blinka/issues/150

Cheers
---David
-- 
Email: druck@druck.org.uk
Phone: +44-(0)7974 108301

gogorz, 2020-05-13:
Adding the user to the `gpio` group fixed the issue, thank you!

closed 2020-05-13 by gogorz