pimoroni/blinkt #88: Docker blues

StevePoling, 2020-08-20:
The Dockerfile doesn't seem to work for me. So, I made some tweaks.
- Changed the FROM image to balenalib/rpi-raspbian:jessie to avoid a "deprecated" message.
- added apt-get install git
- added git clone (this repo)
- dropped the copy of /library and /examples

This allowed me to successfully build a docker image, but when I run it I get this unhappy message:

> Traceback (most recent call last):
  File "larson.py", line 5, in <module>
    import blinkt
  File "/usr/local/lib/python2.7/dist-packages/blinkt.py", line 5, in <module>
    import RPi.GPIO as GPIO
  File "/usr/lib/python2.7/dist-packages/RPi/GPIO/__init__.py", line 23, in <module>
    from RPi._GPIO import *
RuntimeError: This module can only be run on a Raspberry Pi!

Is there something I'm missing to tell python-rpi, or docker that I'm running on an RPi4?

MacOS wouldn't let me attach my Dockerfile. So here it is inline:

> FROM balenalib/rpi-raspbian:jessie
RUN apt-get update -qy && apt-get install -qy \
    git \
    python \
    python-rpi.gpio
ENTRYPOINT []	
RUN git clone https://github.com/pimoroni/blinkt.git /blinkt
WORKDIR /blinkt/library
RUN python setup.py install
WORKDIR /blinkt/examples/
CMD ["python", "larson.py"]

Gadgetoid, 2020-08-20:
I'm afraid the Dockerfile was contributed by @alexellis and I don't have the first clue how to get it working. I'm crossing fingers that invoking his good name will cause him to magically appear and fix all our woes, though!

alexellis, 2020-08-21:
Hi there. I can take a look at getting this working again, but it's going to take me a few days to set up an RPi and find a Blinkt etc.

Gadgetoid, 2023-11-02:
Note: I am deleting the Dockerfile as part of #102 since I have changed *a lot* of things, and don't have the ability to fix it.

Once #102 is merged I would welcome a PR re-adding it, if it's useful. With any luck there wont be another big change like this down the road.

closed 2023-11-02 by Gadgetoid

alexellis, 2023-11-10:
I'd support that decision :smile: