pimoroni/blinkt #91: Support Raspberry Pi OS 64bit

rsun0525, 2021-03-03:
pi@raspberrypi400:~ $ curl https://get.pimoroni.com/blinkt | bash

This hardware is not supported, sorry!
Config files have been left untouched

pi@raspberrypi400:~ $ uname -a
Linux raspberrypi400 5.10.11-v8+ #1399 SMP PREEMPT Thu Jan 28 12:14:03 GMT 2021 aarch64 GNU/Linux

pixelchrome, 2021-04-20:
The manual install of the library works `sudo apt-get install python3-blinkt`

With that I was able to run `rainbow.py` from the [Getting Started with Blinkt!](https://learn.pimoroni.com/tutorial/sandyj/getting-started-with-blinkt)

Gadgetoid, 2022-02-02:
Looks like I'd better get my skates on and allow installs on 64bit - https://www.raspberrypi.com/news/raspberry-pi-os-64-bit/

mahood73, 2022-02-07:
I've just installed this on a Pi 3B, with the official 64 bit OS.

I downloaded the script and removed the two tests that were failing:
arch_check() was reporting it as aarm64, so I changed the code to report it as armhf.
OS_NAME was reported as Debian, so I changed that code to report Raspbian.

The rest of the procedure was unchanged, and it works fine! So hopefully @Gadgetoid doesn't have a lot to do...

Gadgetoid, 2022-02-07:
Thank you @mahood73 that's helpful.

I'm starting to wonder if the OS check is all that much use, since "Debian/aarch64" is pretty generic. It might just need ripping out. I'm tending toward less complicated installer scripts, since these are a legacy holdout from when I had a helping hand working on them and thus a nightmare to maintain.

paulofrazao, 2024-01-30:
Any update on support for Pi 5 and 64bit?

alphanumeric007, 2024-02-02:
WARNING Wall of text to follow!!
Latest Pi OS 64 bit on a Pi **4B**
I got an example to run from command line. 
Thonny just coughed up blink has no attribute... errors.
I have no idea what to do after you reboot to get an example to run?
 
`
pi@raspberrypi:~ $ python3 -m venv ~/my_virtual_env
pi@raspberrypi:~ $ source ~/my_virtual_env/bin/activate
(my_virtual_env) pi@raspberrypi:~ $ sudo apt install python3-dev
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
python3-dev is already the newest version (3.11.2-1+b1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
(my_virtual_env) pi@raspberrypi:~ $ pip install gpiod-2.0.1.tar.gz
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing ./gpiod-2.0.1.tar.gz
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Building wheels for collected packages: gpiod
  Building wheel for gpiod (pyproject.toml) ... done
  Created wheel for gpiod: filename=gpiod-2.0.1-cp311-cp311-linux_aarch64.whl size=93876 sha256=2459abc3eebecfea81ac658c847cd5ce2226fc2188e57ed7d35225d4169f3e19
  Stored in directory: /home/pi/.cache/pip/wheels/88/37/59/bd1f565745b127d16180eb2475aba0b290cc4eaef2c34cde26
Successfully built gpiod
Installing collected packages: gpiod
  Attempting uninstall: gpiod
    Found existing installation: gpiod 2.0.1
    Uninstalling gpiod-2.0.1:
      Successfully uninstalled gpiod-2.0.1
Successfully installed gpiod-2.0.1
(my_virtual_env) pi@raspberrypi:~ $ git clone https://github.com/pimoroni/blinkt -b repackage
Cloning into 'blinkt'...
remote: Enumerating objects: 1425, done.
remote: Counting objects: 100% (243/243), done.
remote: Compressing objects: 100% (137/137), done.
remote: Total 1425 (delta 109), reused 184 (delta 78), pack-reused 1182
Receiving objects: 100% (1425/1425), 411.36 KiB | 3.49 MiB/s, done.
Resolving deltas: 100% (843/843), done.
(my_virtual_env) pi@raspberrypi:~ $ cd blinkt
(my_virtual_env) pi@raspberrypi:~/blinkt $ pip install .
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing /home/pi/blinkt
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Requirement already satisfied: gpiod in /home/pi/my_virtual_env/lib/python3.11/site-packages (from blinkt==0.1.2) (2.0.1)
Requirement already satisfied: gpiodevice>=0.0.2 in /home/pi/my_virtual_env/lib/python3.11/site-packages (from blinkt==0.1.2) (0.0.3)
Building wheels for collected packages: blinkt
  Building wheel for blinkt (pyproject.toml) ... done
  Created wheel for blinkt: filename=blinkt-0.1.2-py3-none-any.whl size=10539 sha256=c3e37203fe386081df943c7264b25ab7e022ed9954080f389f72c5cbbcf14271
  Stored in directory: /tmp/pip-ephem-wheel-cache-h7e295u4/wheels/1f/15/39/e27ca04201b69df7270a606a5d1fc5d9ea73eccd561ad00e2c
Successfully built blinkt
Installing collected packages: blinkt
  Attempting uninstall: blinkt
    Found existing installation: blinkt 0.1.2
    Uninstalling blinkt-0.1.2:
      Successfully uninstalled blinkt-0.1.2
Successfully installed blinkt-0.1.2
(my_virtual_env) pi@raspberrypi:~/blinkt $ pip3 install .
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing /home/pi/blinkt
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Requirement already satisfied: gpiod in /home/pi/my_virtual_env/lib/python3.11/site-packages (from blinkt==0.1.2) (2.0.1)
Requirement already satisfied: gpiodevice>=0.0.2 in /home/pi/my_virtual_env/lib/python3.11/site-packages (from blinkt==0.1.2) (0.0.3)
Building wheels for collected packages: blinkt
  Building wheel for blinkt (pyproject.toml) ... done
  Created wheel for blinkt: filename=blinkt-0.1.2-py3-none-any.whl size=10539 sha256=c3e37203fe386081df943c7264b25ab7e022ed9954080f389f72c5cbbcf14271
  Stored in directory: /tmp/pip-ephem-wheel-cache-ac2aklsj/wheels/1f/15/39/e27ca04201b69df7270a606a5d1fc5d9ea73eccd561ad00e2c
Successfully built blinkt
Installing collected packages: blinkt
  Attempting uninstall: blinkt
    Found existing installation: blinkt 0.1.2
    Uninstalling blinkt-0.1.2:
      Successfully uninstalled blinkt-0.1.2
Successfully installed blinkt-0.1.2
(my_virtual_env) pi@raspberrypi:~/blinkt $ cd examples
(my_virtual_env) pi@raspberrypi:~/blinkt/examples $ python3 cpu_temp.py
`

thirdr, 2025-01-07:
Closed as original issue has been resolved.

closed 2025-01-07 by thirdr