pimoroni/blinkt #101: Pi 5 Support

alessaba, 2023-10-31:
Hi,
I just got a Pi 5 and the first HAT i mounted was a Blinkt i had laying around.
I did a fresh Pi OS install, and ran the installation script

I first got this error
`error: externally-managed-environment`
and the installation stopped, but i installed the apt package and the module seems to show up in python 3

The problem is that when i call the show() function, there is what seems to be a permissions error to /dev/mem.
`Traceback (most recent call last):
  File "/home/pi/Pimoroni/blinkt/examples/rainbow.py", line 22, in <module>
    show()
  File "/usr/lib/python3/dist-packages/blinkt.py", line 76, in show
    GPIO.setup(DAT, GPIO.OUT)
RuntimeError: No access to /dev/mem.  Try running as root!`

Running as root does nothing. There is no error but nothing lights up

i read that it could be something related to /dev/gpiomem and i saw that at least on the Pi 5 (because of the RP1, perhaps?) , there is no gpiomem but just gpiomem0 through gpiomem4, so maybe this could also be the reason, but i have never used /dev/gpiomem manually so i don't remember if it was like this before.

Am i doing something wrong or is there some compatibility problem?

Gadgetoid, 2023-10-31:
> is there some compatibility problem?

Several, in fact, I wont bore you with the details - unless you're in for an otherworldly information dump - but I'm working through our libs and trying to get a functional GPIO environment up and running on the Pi.

Actually Blinkt might be an interesting case. I've dug mine out and will have a look.

alessaba, 2023-10-31:
I am more than curious about the details, if you have the time :)

Otherwise i will have patience and wait while i try other things

Gadgetoid, 2023-10-31:
Okay to save me the typing here's the virtual environment side of things - https://github.com/pimoroni/boilerplate-python/pull/13

And the GPIO side of things - https://lore.kernel.org/linux-gpio/20231025082707.821368-1-phil@gadgetoid.com/

Blinkt has also, as an aside to both of these, suffered from some significant bitrot.

Looks like my Blinkt is now working, though I need to fix clear on exit!

Gadgetoid, 2023-10-31:
Okay this PR has a new working build in - https://github.com/pimoroni/blinkt/pull/102

You will, however, need a pre-release version of libgpiod for Python. I'll post my build here momentarily.

Gadgetoid, 2023-10-31:
Hopefully one of these will work.

Either `sudo apt install python3-dev` and `pip install gpiod-2.0.1.tar.gz`
[gpiod-2.0.1.tar.gz](https://github.com/pimoroni/blinkt/files/13218939/gpiod-2.0.1.tar.gz)

Or unzip this one, and `pip install gpiod-2.0.1-cp311-cp311-linux_aarch64.whl`
[gpiod-2.0.1-cp311-cp311-linux_aarch64.zip](https://github.com/pimoroni/blinkt/files/13218942/gpiod-2.0.1-cp311-cp311-linux_aarch64.zip)

closed 2023-10-31 by alessaba

alessaba, 2023-10-31:
The clear function seems to only work 80% of the time, but every other thing i tested works

Gadgetoid, 2023-10-31:
Thanks. I'll investigate. I'm a little in uncharted territory here, it's also possible the LEDs are just being uncooperative. The protocol is *weird* but software can fix that, too.

Gadgetoid, 2023-10-31:
Well, that's odd! I can't seem to get clear to fail. What code were you running?

alessaba, 2023-10-31:
One of the examples it happened with was rainbow.py

maybe it's something weird with my system, but it’s not that big of a problem