pimoroni/blinkt #103: Developing without the PI for the PI with blinkt

joshy, 2024-01-11:
I couldn't find an answer to the following question. I wanted to develop in python for the pi zero with blinkt on my local computer (mac m1) and then ship the code to the pi. However I can't install blinkt because RPi.GPIO can't be installed. Is this setup somehow supported? 

Thanks for your time and help!
Cheers Joshy

druck13, 2024-01-11:
Have a look at my Linux terminal simulation which allows you to develop blink patterns without a Raspberry Pi.

https://github.com/pimoroni/blinkt/pull/98

I thought it might be useful in situations like yours, but Gadgetoid left the merge request waiting for 2.5 years before choosing to deny it because it might clash with something that still hasn't been done. Very disheartening.

joshy, 2024-02-02:
@druck13 Thanks a lot for the fake_blink! It helped me a lot.

closed 2025-01-09 by thirdr